MicrosoftDocs/windows-driver-docs-ddi #930: Clarify the default parent for WDFSTRING Objects?

PeterGV, 2020-08-13:
Good Morning Ted,

In the super-clever method WdfRegistryQueryMultiString, the Framework creates zero or more WDFSTRINGs representing each of the strings that are read from a given REG_MULTI_SZ value, and inserts each of these String objects into a provided WDFCOLLECTION.

The doc page for WdfRegistryQueryMultiString doesn't say what the default parent is for any created WDFSTRINGs, if the StringAttributes argument is set to NULL.  While the default parent of a WDFSTRING is the WDFDRIVER, in absence of documentation saying so explicitly, the caller can't count on this for strings created internally by WdfRegistryQueryMultiString.  It's entirely possible, for example, that the created strings are parented on the Collection.

I looked up the code to get the definitive answer (so I could give it to you here), but I'll be damned if I can figure it out for sure.

Anyhow, should be a simple change... once one of the devs who owns this tells us what the parent actually is, of course.

Thanks!!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3dda0c02-d50e-f993-eb34-1fd1ff5c7654
* Version Independent ID: 5729e82c-5855-d920-dad0-cf186fba4cbb
* Content: [WdfRegistryQueryMultiString function (wdfregistry.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfregistry/nf-wdfregistry-wdfregistryquerymultistring)
* Content Source: [wdk-ddi-src/content/wdfregistry/nf-wdfregistry-wdfregistryquerymultistring.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfregistry/nf-wdfregistry-wdfregistryquerymultistring.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-08-17:
Hey Peter, always good to hear from you. Turns out the parent is the WDFDRIVER. I just pushed an update to add this to the Remarks of the doc page. That should be live within the hour.

BTW, as of this weekend, SAL annotations are back on the WDK DDI doc pages! Let me know if this is useful :)

closed 2020-08-17 by tedhudek

PeterGV, 2020-08-17:
>SAL annotations are back on the WDK DDI doc pages! Let me know if this is useful 

OF COURSE it's useful.  Love it, in fact.  THANK you, to whoever made this happen.

EliotSeattle, 2020-08-17:
Thanks, @PeterGV! 
It was a team effort, but it was mostly led by @tedhudek and @DOMARS, along with friends on our docs build team. We're still getting some of the details taken care of, but we've got a working plan to get back to parity.
Eliot