MicrosoftDocs/windows-driver-docs-ddi #941: PFND3D11DDI_SETRENDERTARGETS callback function feedback questions

ChuckM2, 2020-08-25:
On this page:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3d10umddi/nc-d3d10umddi-pfnd3d11ddi_setrendertargets

1. UAVIndex is mentioned only once, but it's not clear what it is. Should this be UAVStartSlot?

2. UAVNumberUpdated is mentioned only once, but it's not clear what it is.  Should it be UAVRangeSize or NumUAVs?

3. UAVFirsttoSet is mentioned only once, but it's not clear what it is.  Should it be UAVRangeStart?

4. UAVRangeStart is unclear to me, but maybe once the Note below is clarified it will make more sense.  Is UAVRangeStart the first offset in the phUnorderedAccessView array that has been updated?  Is UAVRangeStart an offset from UAVStartSlot or relative to the start of phUnorderedAccessView?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a6967b37-a88d-69f3-4a31-af9b77930d14
* Version Independent ID: 3d17709f-3b60-d9df-f7ae-3e97e268dd02
* Content: [PFND3D11DDI_SETRENDERTARGETS (d3d10umddi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3d10umddi/nc-d3d10umddi-pfnd3d11ddi_setrendertargets)
* Content Source: [wdk-ddi-src/content/d3d10umddi/nc-d3d10umddi-pfnd3d11ddi_setrendertargets.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3d10umddi/nc-d3d10umddi-pfnd3d11ddi_setrendertargets.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2020-10-27:
@ChuckM2 , thanks for reporting the param issues and asking great questions about them. I've updated the page to fix param names and added lots of technical goodies to make it clearer.

closed 2020-10-27 by lorihollasch

ChuckM2, 2020-10-28:
> @ChuckM2 , thanks for reporting the param issues and asking great questions about them. I've updated the page to fix param names and added lots of technical goodies to make it clearer.

Thank you for the update.  A few more questions.

>"To choose the whole range (no change believed to happen), UAVRangeStart = 2 and UAVRangeSize = 3."

Q1. Why does UAVRangeStart/Size indicate all 3 elements if "no change" happened?  Would UAVRangeSize ever be 0?

>https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3d10umddi/nc-d3d10umddi-pfnd3d11ddi_setrendertargets

Q2. In this example, which Arg6/phUnorderedAccessView array elements are accessed/valid?  From 0 to NumUAVs, regardless of UAVStartSlot?   
Q3. Again looking at phUnorderedAccessView array accesses, do the modified UAV elements start at array offset (UAVRangeStart - UAVStartSlot) to  (UAVRangeStart - UAVStartSlot + UAVRangeSize)?

ChuckM2, 2020-10-28:
Maybe add in parameters/remarks that "UAVRangeStart" is a slot and will never be less than UAVStartSlot.