MicrosoftDocs/windows-driver-docs-ddi #949: mistake on NtQuerySecurityObject page

maharmstone, 2020-09-10:
There's a mistake on this page - it says that the SecurityInformation argument is a pointer, when actually it's a ULONG and thus passed by value.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 31063e0f-7395-ae8f-d700-aebadeca7527
* Version Independent ID: db6924d6-9530-961d-16d4-52969ae79600
* Content: [NtQuerySecurityObject function (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ntquerysecurityobject)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ntquerysecurityobject.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntquerysecurityobject.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

maharmstone, 2020-09-10:
There's also a contradiction - it says that ``SecurityDescriptor`` will be a relative SD, but then in the first sentence of Remarks mentions absolute SDs.

hsebs, 2020-10-16:
First sentence of Remarks describes SECURITY_DESCRIPTOR.

lorihollasch, 2020-10-21:
Thanks again, @hsebs , this is also now fixed!

closed 2020-10-21 by lorihollasch