MicrosoftDocs/windows-driver-docs-ddi #951: Incorrect documentation

Shyamal-V, 2020-09-15:
This is the WDM header for kernel mode drivers. But the documentation incorrectly shows UMDF's documentation for this.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7d843ac3-43f8-d4c7-2735-c2488b0abd03
* Version Independent ID: 0d997e2c-7d15-5642-d0b2-2095430175a4
* Content: [WRITE_REGISTER_ULONG64 function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-write_register_ulong64)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-write_register_ulong64.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-write_register_ulong64.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-09-15:
Thanks for the report @Shyamal-V, looks like this is also true for:

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-ppower_action
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-read_register_buffer_ulong64
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-read_register_ulong64

Working on a fix.

tedhudek, 2020-09-16:
These fixes now appear on the live site.

Thanks again for letting us know.

closed 2020-09-16 by tedhudek