MicrosoftDocs/windows-driver-docs-ddi #962: One or more IMiniportWaveRTStream interface methods missing from overview/index

mr-intj, 2020-09-27:
I reported this for a different URL under docs.microsoft.com; but this appears to be an issue in _multiple places_ (I've been clicking on search result links, and haven't paid close attention to the URL path).
 
In portcls.h, the **IMiniportWaveRTStream** interface is declared as having eight methods:

- AllocateAudioBuffer
- FreeAudioBuffer
- GetClockRegister
- GetHWLatency
- GetPosition
- GetPositionRegister
- SetFormat
- SetState

In this case, I believe the **methods** overview page lists only 7 out of 8 interface methods. The index (in the pane on the far left) lists only 6 out of 8.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f88bb5eb-1633-d0b5-0e7a-a86c429739ad
* Version Independent ID: cba05873-e842-50bf-cdab-d569c2009aba
* Content: [IPortWaveRTStream (portcls.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/portcls/nn-portcls-iportwavertstream)
* Content Source: [wdk-ddi-src/content/portcls/nn-portcls-iportwavertstream.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/portcls/nn-portcls-iportwavertstream.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-09-29:
@mr-intj - Yes it looks like there are issues here with all of the methods being listed and rendered. I will investigate.

DOMARS, 2020-10-02:
@mr-intj - Thanks for this issue. I have restored SetFormat and SetState to  IMiniportWaveRTStream so that all 8 of the 8 methods are now present.

closed 2020-10-02 by DOMARS