MicrosoftDocs/windows-driver-docs-ddi #973: Truncated paragraph + contradictory statement(s)

mr-intj, 2020-10-11:
**Remarks** section:

> The actual size must be at least the requested size; otherwise, the Audio
 
Text at end of paragraph cuts off mid-sentence ^^^

> ...the actual size of the buffer is the requested size **rounded (up or down)** to the nearest sample or other hardware-constrained boundary. The actual size **must be at least the requested size**; otherwise, ...

This is contradictory. If the actual size must be at least the requested size, then rounding _down_ is never an option. My experience seems to indicate that returning a value _larger than_ the requested size (e.g., due to rounding _up_) is also rejected.

This would only seem to make sense if the rounding was being done by the caller, but the text above says `ActualSize` (which is a return value) is equal to `RequestedSize` (which is a parameter) rounded... -- so the implication is that this is to be done in the method implementation by the callee, and not by the caller, prior to making the call.

Meanwhile (at least for WaveRT), returning a value that is smaller is a documented no-no, and empirical evidence suggests that returning a value that is larger results in rejection. Something doesn't seem to add up here.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ab27fe16-85eb-4695-49bc-631514547040
* Version Independent ID: 74929cef-44a6-84da-32a1-795865016201
* Content: [IMiniportWaveRTStream::AllocateAudioBuffer (portcls.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/portcls/nf-portcls-iminiportwavertstream-allocateaudiobuffer)
* Content Source: [wdk-ddi-src/content/portcls/nf-portcls-iminiportwavertstream-allocateaudiobuffer.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/portcls/nf-portcls-iminiportwavertstream-allocateaudiobuffer.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ab27fe16-85eb-4695-49bc-631514547040
* Version Independent ID: 74929cef-44a6-84da-32a1-795865016201
* Content: [IMiniportWaveRTStream::AllocateAudioBuffer (portcls.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/portcls/nf-portcls-iminiportwavertstream-allocateaudiobuffer)
* Content Source: [wdk-ddi-src/content/portcls/nf-portcls-iminiportwavertstream-allocateaudiobuffer.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/portcls/nf-portcls-iminiportwavertstream-allocateaudiobuffer.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-12-21:
@mr-intj - Yes it does look like the text was truncated. I will work to correct this topic. 

The paragraph should read:

The audio device might require the audio buffer to begin and end on sample boundaries or to meet other types of hardware-dependent alignment constraints. If sufficient memory is available, the actual size of the buffer is the requested size rounded (up or down) to the nearest sample or other hardware-constrained boundary. The actual size must be at least the requested size; otherwise, the Audio Session API (WASAPI) audio engine won't use the buffer, and stream creation will fail.

DOMARS, 2020-12-22:
@mr-intj - I restored the missing text, thanks for pointing out that the text was truncated. For your question on the allocate size, I would look at the sysvad sample miniwavestream.cpp starting on line 778. This code uses this technique for the block size.   `RequestedSize_ -= RequestedSize_ % (m_pWfExt->Format.nBlockAlign);`  For additional questions, two groups to consider. The OSR dev forums at https://community.osr.com/ and we recently moved the MSDN Forums to this location for Windows development. https://docs.microsoft.com/answers/products/windows. Good luck with your WDM audio driver development work.

closed 2020-12-22 by DOMARS