MicrosoftDocs/windows-driver-docs-ddi #982: Can't find SB16 audio driver sample code

mr-intj, 2020-10-21:
> An adapter driver that overwrites one or more of the pointers above with a pointer to its own IRP handler can call PcDispatchIrp from within its handler routine in order to forward the IRP to PortCls. For a code example, see the SB16 sample audio driver in the Microsoft Windows Driver Kit (WDK).

I can't find any trace of this "SB16" sample online or in the WDK releases going back to WIndows 8.0.

Unfortunately, right now I **_really need_** an example of a miniport driver handling and forwarding IRPs to PortCls. :-(

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fff7cf7b-a1e0-3b3f-d7a0-fe850f24a2db
* Version Independent ID: 724a9e9a-b562-5241-382e-5ee9774e8373
* Content: [PcInitializeAdapterDriver function (portcls.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/portcls/nf-portcls-pcinitializeadapterdriver)
* Content Source: [wdk-ddi-src/content/portcls/nf-portcls-pcinitializeadapterdriver.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/portcls/nf-portcls-pcinitializeadapterdriver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-10-29:
@mr-intj - Thanks for filing this issue. At one point the older samples where available for download. I will look at the SB16 references.

It looks like in the current day Sysvad sample, towards the end of the adapter.cpp, as part of the PnP IRP handler, is this code:

 `ntStatus = PcDispatchIrp(_DeviceObject, _Irp);`

https://github.com/microsoft/Windows-driver-samples/tree/master/audio/sysvad

But this code may not include the forwarding part of what you are after.

This topic describes the use of the IRQL level, which much be carefully determined and set.
https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/dispatch-routines-and-irqls

DOMARS, 2020-10-29:
@mr-intj - Another possible resource for older samples would be the Windows 7 WDK download - https://www.microsoft.com/en-us/download/confirmation.aspx?id=11800

DOMARS, 2020-12-21:
@mr-intj - I believe that the SB16 sample should be included in the 7.0  WDK ISO download - https://www.microsoft.com/en-us/download/confirmation.aspx?id=11800. Another resource for development work is the brand new driver development forums available here: https://docs.microsoft.com/answers/products/windows. Good luck with your WDM audio driver development work.

closed 2020-12-21 by DOMARS