MicrosoftDocs/windows-driver-docs-ddi #990: Given the historical confusion around 'sample size', please be more explicit

mr-intj, 2020-11-05:
> `SampleSize`
> 
> Specifies the sample size of the data, for fixed sample sizes, or zero, if the format has a variable sample size.

I'm guessing that the units here are **bytes** and not **bits**, but it might be good to make that more explicit here.

Also, there is some legacy confusion between the terms _sample_ and _frame_ (a single frame of audio, which contains a sample for each channel).

Finally, there is additional ambiguity at times where sample size (bits-per-sample) are discussed. In some cases, what's actually being referred to is the size of the sample _container_, which may contain a padded sample. For example, a 20-bit sample may be padded within a 24-bit sample container...

Cheers.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 50e0f26d-1cce-f0c2-5a22-8df1707a6612
* Version Independent ID: 9fb013df-0e36-6059-b485-f283123c6a3f
* Content: [KSDATAFORMAT (ks.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ks/ns-ks-ksdataformat)
* Content Source: [wdk-ddi-src/content/ks/ns-ks-ksdataformat.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ks/ns-ks-ksdataformat.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-01-27:
Adding DOMARS as a reviewer on this issue as this appears to be audio related.

barrygolden, 2021-01-29:
@mr-intj, thanks for pointing out the ambiguity around the SampleSize member description.

I'm adding the feedback you've provided to an internal work item to correct and update the KSDATAFORMAT structure topic.

closed 2021-01-29 by barrygolden