MicrosoftDocs/windows-driver-docs-ddi #993: NtCreateResourceManager's RmGuid is not optional

maharmstone, 2020-11-17:
I think RmGuid has been marked as optional in error - I get a STATUS_ACCESS_VIOLATION BSOD if it's NULL, called from kernel space. The offending instruction is at tm!NtCreateResourceManagerExt+0x144.

This is on Windows 10 1511 on x86 - I've not checked on later versions or on amd64. Calling it from userspace returns STATUS_ACCESS_VIOLATION rather than crashing.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d109597a-f18b-a0e0-9b87-59120b41fb22
* Version Independent ID: f303f3a6-7bfc-fd1a-6992-dc66673d8a10
* Content: [NtCreateResourceManager function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-ntcreateresourcemanager)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-ntcreateresourcemanager.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-ntcreateresourcemanager.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

maharmstone, 2020-11-17:
Just checked - the definition in wdm.h is marked _In_ rather than _In_opt_, so this appears to be purely an issue with the documentation.

tedhudek, 2020-12-02:
Great find @maharmstone, thanks.

Just pushed the doc change, should appear on the live site within an hour.

closed 2020-12-02 by tedhudek