MicrosoftDocs/windows-driver-docs-ddi #994: Missing and/or Incorrcet Information

MartinDrab, 2020-11-17:
* `ExDesktopObjectType ` is supported starting Windows 10 but permissions we are allowed to remove are not listed.
* it seems that the table with removable process rights does not list all permissions we can actually remove (at newest Windows versions at least), e.g. `PROCESS_VM_READ` is not listed in the `RetainAccess` member of the `OBJECT_TYPE` structure, only `SYNCHRONIZE`, and `PROCESS_QUERY_LIMITED_INFORMATION`.

I did not checked the table with removable thread rights.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ba87921a-3667-3d45-d4a4-8455666ea2bd
* Version Independent ID: 112fb965-6770-205e-0766-6a8d7d8efb50
* Content: [_OB_PRE_CREATE_HANDLE_INFORMATION (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_ob_pre_create_handle_information)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_ob_pre_create_handle_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_ob_pre_create_handle_information.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-12-03:
Hi @MartinDrab, 

Thanks for the feedback.  When you say "the table with removable process rights" do you mean in the description of the `DesiredAccess` member of https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_ob_pre_create_handle_information?

Also, I'm not finding a doc page that references the `RetainAccess` member of the **OBJECT_TYPE** structure.  Do you have a link for that?

MartinDrab, 2020-12-03:
>  When you say "the table with removable process rights" do you mean in the description of the DesiredAccess member of https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_ob_pre_create_handle_information?

Yes, I do.

> Also, I'm not finding a doc page that references the RetainAccess member of the OBJECT_TYPE structure. Do you have a link for that?

Well, `RetainAccess` member is not officially documented (no member of the `OBJECT_TYPE` structure is), however, its purpose is kinda known within the kernel dev community -- it contains access rights that cannot be removed by OB callbacks. 

What I am suggesting is that somebody should check whether all rights that can be removed from the `DesiredAccess` mask are really listed within the table(s), since it seems that there are more of them on newer versions of Windows.

tedhudek, 2020-12-11:
Hi @MartinDrab 

Thanks again for reporting this.  It turns out that `ExDesktopObjectType` is actually not supported.  We will update this.

Otherwise, the info on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_ob_pre_create_handle_information is correct.

closed 2020-12-11 by tedhudek