MicrosoftDocs/windows-driver-docs-ddi #995: Inconsistent description of memory management with NdisInitializeString

I-gor-C, 2020-11-18:
The documentation states:

> `Destination`
> 
> A pointer that is NULL when NdisInitializeString is called; on return from this function, this parameter points to an NDIS_STRING type that describes an initialized counted string.

That implies that `NdisInitializeString` allocates the entire `NDIS_STRING` structure.

Later it is stated:

> The buffer allocated by `NdisInitializeString` should be released with the `NdisFreeString` function.

Looking at `NdisFreeString` implementation in WDK we see this:

`#define NdisFreeString(String) NdisFreeMemory((String).Buffer, (String).MaximumLength, 0)`

I.e., it frees only the internal storage buffer of `NDIS_STRING`.
So what happens with the entire NDIS_STRING structure that is presumably allocated by `NdisInitializeString` per quote above?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: eb2fbe4c-782e-a837-ed88-d05e79720d9a
* Version Independent ID: 94357920-2f35-7728-efee-bf5958fdae7c
* Content: [NdisInitializeString function (ndis.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ndis/nf-ndis-ndisinitializestring)
* Content Source: [wdk-ddi-src/content/ndis/nf-ndis-ndisinitializestring.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ndis/nf-ndis-ndisinitializestring.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2020-11-18:
Thanks @I-gor-C for opening this issue. This enquiry will take some time to research, and then we'll get back to you.

jtippet, 2020-11-18:
Thanks for catching this error.  The description for `Destination` *should* say something like:

> This parameter points to an NDIS_STRING with a NULL Buffer.  On return from this function . . . .

aviviano, 2020-11-19:
Thanks again @I-gor-C. The description for the Destination parameter is now updated on the site.

closed 2020-11-19 by aviviano