MicrosoftDocs/windows-driver-docs-ddi #996: Values for the PC_EXIT_LATENCY enumeration differ between pages

darugeri, 2020-11-19:
The table listing the values for the `NewD3ExitLatency` parameter differs from the values presented in the linked page for `PC_EXIT_LATENCY` [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/portcls/ne-portcls-_pc_exit_latency).

I'm not sure which set is the correct one, but we can sync offline if needed.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7d2575e8-1747-2b14-a7b4-0fa04321f6f7
* Version Independent ID: 7e6f55e6-af4e-663e-5f62-e69aac1f2720
* Content: [IAdapterPowerManagement3::D3ExitLatencyChanged (portcls.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/portcls/nf-portcls-iadapterpowermanagement3-d3exitlatencychanged)
* Content Source: [wdk-ddi-src/content/portcls/nf-portcls-iadapterpowermanagement3-d3exitlatencychanged.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/portcls/nf-portcls-iadapterpowermanagement3-d3exitlatencychanged.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-12-22:
@darugeri - I will follow up in email. Now that I look at this, I don't follow if NewD3ExitLatency is sort of redefining with these new values, or if we need to somehow create a new enum for  D3ExitLatency to have these new values. The enum itself of course just has the values of 0,1 and 2.

DOMARS, 2021-01-21:
@darugeri - Following up in email to see if we can get this figured out.   :)

DOMARS, 2021-02-04:
@darugeri - Following up in email to see if we can move this forward.

DOMARS, 2021-03-12:
@darugeri - Following up in email. I converted this git issue to an internal work item, so closing this issue. Happy to make whatever updates to the doc you would like,

closed 2021-03-12 by DOMARS