MicrosoftDocs/windows-driver-docs-ddi #998: FSCTL_SET_REPARSE_POINT_EX undocumented

maharmstone, 2020-11-19:
Unless I'm missing something, FSCTL_SET_REPARSE_POINT_EX is undocumented, even though REPARSE_DATA_BUFFER_EX is. The links on this page go to FSCTL_SET_REPARSE_POINT instead.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 15aae2b6-02ff-6cff-3dff-c26b805660cc
* Version Independent ID: 4eaeb721-4dc1-eb31-b8cf-8454154022a5
* Content: [_REPARSE_DATA_BUFFER_EX (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_reparse_data_buffer_ex)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_reparse_data_buffer_ex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_reparse_data_buffer_ex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

KalleOlaviNiemitalo, 2020-11-27:
The [FSCTL_SET_REPARSE_POINT_EX control code](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/fsctl-set-reparse-point-ex) documentation appears to be live now.

maharmstone, 2020-11-28:
So it is - thanks @lorihollasch.

lorihollasch, 2020-12-02:
Thanks for reporting the issue, @maharmstone !

closed 2020-12-02 by lorihollasch