MicrosoftDocs/windows-driver-docs-ddi #999: Typo in function name

I-gor-C, 2020-11-21:
Named `KeLowerIrql~r1` instead of `KeLowerIrql`

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f906dc40-5535-9fa6-e9ea-66828417e308
* Version Independent ID: 99b9bdd2-7aff-8523-3e82-92091fac3e0f
* Content: [KeLowerIrql function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-kelowerirql?source=docs)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-kelowerirql.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-kelowerirql.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-12-02:
Thanks @I-gor-C for the report.  We're working on this one - I'll keep you updated.

tedhudek, 2020-12-11:
Hi again @I-gor-C,

This one looks like it might take a couple months to fix. There are ~28 of these across the WDK API docs.  It's a known issue, and there's an internal feature request tracking it.

As such, I'm going to close out this doc issue.  Apologies for the visual clutter!

Thanks again for bringing it to our attention.

closed 2020-12-11 by tedhudek