MicrosoftDocs/windows-driver-docs-ddi #1002: Misleading wording in Remarks section

wmjdgla, 2020-11-26:
The second paragraph of the Remarks section states:

> A successful call to FltSetStreamContext increments the reference count on NewContext. When the context pointed to by NewContext is no longer needed, the minifilter must call FltReleaseContext to decrement its reference count.

This seems to imply that FltReleaseContext decrements the reference count that was incremented by FltSetStreamContext. The next paragraph proceeds to state that on failure, calling FltReleaseContext decrements the reference count that was incremented by FltAllocateContext. However the Remarks section of FltSetFileContext and FltSetInstanceContext makes no such distinction, and simply states that FltReleaseContext must be called regardless of whether FltSetXxxContext succeeds.

Cross referencing the [Releasing Contexts](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/releasing-contexts) page:

> Every successful call to one of the following routines must eventually be matched by a call to FltReleaseContext: ...

but no FltSetXxxContext functions are in the list. It goes on to state that

>  FltReleaseContext is called regardless of whether the call to FltSetInstanceContext succeeds. In both cases, the caller must call FltReleaseContext to release the reference set by **FltAllocateContext (not FltSetInstanceContext)**.

Hence may I suggest that the wording of the paragraph be amended so as to make it clear that calling FltReleaseContext regardless of failure is meant to decrement the reference count incremented by FltAllocateContext? It would also be good for the wording to be standardized across the Remarks section of all the FltSetXxxContext pages (the FltSetStreamHandleContext page uses the same misleading wording).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 58cba283-2c55-8dc7-b69c-70b7d3d89349
* Version Independent ID: d7e203a9-59bc-1ecb-3a17-afc2484c3ee1
* Content: [FltSetStreamContext function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltsetstreamcontext)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltsetstreamcontext.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltsetstreamcontext.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

hsebs, 2021-01-08:
As far as I know, references incremented by FltSetXxxContext are decremented when the object is cleaned up.

See [Referencing Contexts](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/referencing-contexts)

lorihollasch, 2021-01-30:
@wmjdgla , thanks for this issue - the reference counting info was incomplete on a lot of context-related pages. I've updated every relevant context-related refpage and also added details to the conceptual material (I added links to conceptual material on every context-related refpage as well).

closed 2021-01-30 by lorihollasch