MicrosoftDocs/windows-driver-docs-ddi #1004: Why EmptyWorkingSet() can swap out pages locked by MmProbeAndLockPages()

zyimin, 2020-11-28:
According to the docs, MmProbeAndLockPages() can lock pages. But I found that EmptyWorkingSet() can swap out the locked pages. It has becomed modified status by looking up from rammap. Is there any other routine that I can have the page really locked?

EliotSeattle, 2020-11-30:
Content in question is [https://docs.microsoft.com/windows-hardware/drivers/ddi/wdm/nf-wdm-mmprobeandlockpages](https://docs.microsoft.com/windows-hardware/drivers/ddi/wdm/nf-wdm-mmprobeandlockpages) in wdm.h

tedhudek, 2020-12-03:
Hi @zyimin, thanks for the feedback.  I've just pushed two commits to address your question, please see 27e3140 and 74ffd5b.

These changes should appear on docs.microsoft.com within an hour.

zyimin, 2020-12-04:
> Hi @zyimin, thanks for the feedback. I've just pushed two commits to address your question, please see [...]

Thanks a lot! And I have another question. Is there any mechanism that page swapped notification? For example, in linux, kvm has a mmu sync mechanism so we can sychronize ept tables if the guest page is being swapped out.

tedhudek, 2020-12-08:
Hi @zyimin, no, sorry, there is no such mechanism.

zyimin, 2021-02-20:
> Hi @zyimin, no, sorry, there is no such mechanism.

Hi tedhudek,

I have to reopen this topic. I have taken tests recently, and found that the physical page can be actually locked on page frame by calling MmProbeAndLock(). But there's one thing still confusing me. If I don't call VirtualLock() in userspace application, the virtual address page can also be touched even if it might be swapped out. So I want to ask whether it can be mapped to the original physical page via page fault if I didn't free the virtual address or remapped it to others or freed it.

Thansk very much!

Yi Min

zyimin, 2021-02-23:
As the test result, I found the vaddr of the virtual page would be modified after issue EmptyWorkingSet(). I didn't call VirtualLock() for the vaddr, but called the kernel driver via IOCTL and the driver issued MmProbeAndLock() for the vaddr(user mode). After the vaddr changed to be modified, if the code touched the vaddr then I saw it was remapped to the original physical addr via RAMMAP tool and I guess the physical page has never been swapped out even though it was modifed status once. My understanding is the vaddr can be remapped to the original physcial addr and the physical page cannot be swapped out even if EmptyWorkingSet() called. The last question is about that task manager can't reflact the real usage of the process. The kernel driver locked the userspace process' memory via MmProbeAndLock() and the locked address space can still become inactive state. The inactive virtual address would not be counted as the used memory showed in task manager but its mapped physical page are still existing. Not sure whehter my understanding is right. If wrong, please correct me. Expect your response. Thansk very much!

tedhudek, 2021-02-24:
Using VirtualLock means the working set trimmer will not trim your *usermode virtual address* (because you're trying to eliminate exposure to a potential page fault - can be soft or hard – and therefore takes longer).  However, VirtualLock does *NOT* guarantee that the page won’t be traded or temporarily (this is invisible to the app) be made inaccessible (for page scrubbing etc).

Using MmProbeAndLockPages applies a reference count to the page, thus ensuring the page cannot be reused until the driver also calls MmUnlockPages.  Regardless of what happens to virtual mappings like above.  This makes it safe for a driver to DMA into the page etc.  Warning: doing this will pockmark memory badly, making many other things inefficient and potentially unstable (available pages will plummet causing significant aging/trimming pressure, we cannot quarantine bad pages, we cannot build larger pages, etc).

closed 2021-02-24 by tedhudek

zyimin, 2021-02-26:
Thanks for your kind answer!