MicrosoftDocs/windows-driver-docs-ddi #1005: Reference to non-existing field

I-gor-C, 2020-12-08:
The page has the following piece:

> `PcwRegister` may return `STATUS_INVALID_PARAMETER_2` in the following cases:
> * ...
> * The `Info->Flags` field contains a value not recognized by the running version of Windows.
> 

However, there is no `Flags` in the `Info` struct, which is defined like this in `wdm.h`:
```
typedef struct _PCW_REGISTRATION_INFORMATION {
    _In_ ULONG Version;
    _In_ PCUNICODE_STRING Name;
    _In_ ULONG CounterCount;
    _In_reads_(CounterCount) PPCW_COUNTER_DESCRIPTOR Counters;
    _In_opt_ PPCW_CALLBACK Callback;
    _In_opt_ PVOID CallbackContext;
} PCW_REGISTRATION_INFORMATION, *PPCW_REGISTRATION_INFORMATION;
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 12252677-9e7c-fa24-2e42-b1d2b9e69c69
* Version Independent ID: e6482801-e99d-323e-e0a5-03ae9473f8dd
* Content: [PcwRegister function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-pcwregister)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-pcwregister.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-pcwregister.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-12-11:
Hi @I-gor-C 

Thanks for the report.  It turns out that the `Flags` field is present when building with a Manganese kit - `NTDDI_VERSION >= NTDDI_VERSION_MN` (19645).  That came out around June, which you can see at the [Flight Hub](https://docs.microsoft.com/en-us/windows-insider/flight-hub/) site.

I'll update the doc.

I-gor-C, 2020-12-11:
Hi @tedhudek

Thanks for checking this out.
I see a lot of references in the docs to `NTDDI_VERSION_MN` and `NTDDI_VERSION_FE`. These codenames aren't known and aren't comparable to known OS versions. So when I see in a doc something like `for version >= NTDDI_VERSION_MN` - there is not much I can squeeze out of this piece of information. Having instead something like `version >= NTDDI_VERSION_WIN10_19645` would be much more useful.

tedhudek, 2020-12-11:
That's good feedback.  I see that the `NTDDI_VERSION` defines are in `sdkddkver.h` in the SDK, but the defines just look like:

`#define NTDDI_WIN10_VB                      0x0A000008  /* ABRACADABRA_WIN10_VB */`

So no build number.

@EliotSeattle - one option would be to add a table of NTDDI_VERSION to build number mappings to https://docs.microsoft.com/en-us/windows-hardware/drivers/what-s-new-in-driver-development

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_pcw_registration_information now has Flags btw

EliotSeattle, 2020-12-11:
@tedhudek The version table is a **great** idea. Is What's New the right visibility? Also, the SDK has https://docs.microsoft.com/en-us/windows/win32/winprog/using-the-windows-headers, which describes the same info (but needs an update). Should we promote that in our content instead?

tedhudek, 2020-12-12:
@I-gor-C, for the example of https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_pcw_registration_information, if you wanted to use the Flags member, would you need to target Manganese with your driver?  Or could you target Vibranium and it's enough to simply have a WDK that includes this support?

In other words, what would you do with the info that you need 19645 or above for this?  Knowing that will help me figure out the best way to add the info to the docs

I-gor-C, 2020-12-12:
@tedhudek I would assume that as long as builds numbers are public knowledge, knowing what specific build number is required should be enough. I suppose the decision what to do with this info is vendor-specific and depends on the expected target OS version(s).

tedhudek, 2020-12-15:
Since it's only the PCW support (that I know of) and that functionality has a build number on the doc page, we decided to leave this as-is. Thanks again for the feedback @I-gor-C!  Happy holidays

closed 2020-12-15 by tedhudek