MicrosoftDocs/windows-driver-docs-ddi #1011: There Should be a WDF_REQUEST_SEND_OPTIONS_ASYNCHRONOUS

PeterGV, 2020-12-11:
I *know* this isn't a doc issue... but I'm going to put it here anyways.

It's quite silly that there's not a WDF_REQUEST_SEND_OPTIONS_ASYNCHRONOUS flag defined.  Yes, I *know* this is the default option.  But having options for ALL the send options (sync, send-and-forget, AND async) would make everything tidy and consistent.  Yes, I want to still be able to leave it out, as we do today... but making these things special cases makes it so much harder for people to learn.

Even if the dev owner wants to #define the ASYNC option as zero... that'd be helpful.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1a2d7cbc-b63f-553a-71da-fe22c3af322b
* Version Independent ID: 925bd7b1-284d-1320-1648-ba823f76187d
* Content: [_WDF_REQUEST_SEND_OPTIONS_FLAGS (wdfrequest.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfrequest/ne-wdfrequest-_wdf_request_send_options_flags#feedback)
* Content Source: [wdk-ddi-src/content/wdfrequest/ne-wdfrequest-_wdf_request_send_options_flags.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfrequest/ne-wdfrequest-_wdf_request_send_options_flags.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

EliotSeattle, 2020-12-11:
Thanks, @PeterGV. @tedhudek will forward to the right people. 
Appreciate the insight and feedback!!!
Eliot

PeterGV, 2020-12-11:
Its embarrassing that I've been teaching people this material for 15 years now, and every single time I teach people about WdfRequestSend I say to myself "I should tell somebody that this should be consistent, and there should be a #define for _ASYNC" ... and for 15 years I've never remembered to mention it to anyone.  

It's even more embarrassing that I wind-up mentioning it here (and asking Ted to ask the dev owner to do this for me).  The sad fact is I'm not sure who the lead is for WDF anymore :-(

EliotSeattle, 2020-12-11:
"Sometimes the winding road is the best path to the mountain top."

tedhudek, 2020-12-11:
Hey Peter, thanks as always for the feedback.  We do appreciate it and take action where possible.

I've passed along your feedback to the right folks.  Since there is no doc impact here, I'm going to close this one out.  

Happy holidays!

closed 2020-12-11 by tedhudek