MicrosoftDocs/windows-driver-docs-ddi #1013: Flags field lacks important values

OZ1, 2020-12-14:
Flags field is described only for IRP_MJ_WRITE, however there are more important cases such as SL_FORCE_ACCESS_CHECK for IRP_MJ_CREATE. Please document more this field.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c1b1c00f-1701-5665-cdeb-6120305abc72
* Version Independent ID: 2f6bf8f4-3e19-e4e1-36f4-fcaa8e40c467
* Content: [_IO_STACK_LOCATION (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_io_stack_location)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_io_stack_location.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_io_stack_location.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-12-14:
Thanks for the feedback, @OZ1 

I do see SL_FORCE_ACCESS_CHECK mentioned on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/ns-fltkernel-_flt_io_parameter_block.  Is it just that Flag you would like added to https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_io_stack_location, or are there others?

tedhudek, 2020-12-15:
I added a link from the Flags description of https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_io_stack_location#members to https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/irp-mj-create.  That change will appear on the live site within an hour. 

The `SL_*` flags were documented on that page as a result of https://github.com/MicrosoftDocs/windows-driver-docs/issues/2279.

Thanks again for the feedback.

closed 2020-12-15 by tedhudek