MicrosoftDocs/windows-driver-docs-ddi #1014: Some words about external serialization would help.

rodwiddowson, 2020-12-14:
[Enter feedback here]
This page (and all the others about the functions which establish oplocks in the FsRtl and Flt namespaces) could do with some words about the synchronization that thecaller is required to provide.  If you contact me back channel I can provide some input which I harvested from a study of the published [fat sources](https://github.com/microsoft/Windows-driver-samples/tree/master/filesys/fastfat).

If you will forgive the [RFC2119](https://tools.ietf.org/html/rfc2119)-speak it should be something like

Filters and File Systems which call this function **MUST** synchronize (lock exclusive) calls which create oplocks for the same FCB.  These are 
- IRP_MJ_CREATE with the `FILE_OPEN_REQUIRING_OPLOCK` bit set in Options
- IRP_MJ_FILE_SYSTEM_CONTROL for the "oplock controls".

(This stops the oplock package from crashing - again contact me back channel and I may be able to help with an example.  Or not)

Filters and File Systems which call this function **MAY** also synchronize the above calls with any calls which break oplocks.  This may provide better seralization semantics.

In [this note](https://community.osr.com/discussion/248984/who-is-responsible-for-synchronizing-access-to-the-oplock-structure), Christian seems to indicate that the second is a **RECOMMENDED** but I would argue strongly that this is not the case:   if one application establishes an oplock at the same time as something which might break it, no internal serialization by the FSD can mitigate against the IRPs arriving in any order.  Further FAT most definitely doesn't do this.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 379727d5-1bb2-c31b-cb8d-6a43542c7007
* Version Independent ID: 19dc6b3f-c6f2-a0eb-c612-695a0fa8bae4
* Content: [FsRtlOplockFsctrl function (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-_fsrtl_advanced_fcb_header-fsrtloplockfsctrl)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-_fsrtl_advanced_fcb_header-fsrtloplockfsctrl.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-_fsrtl_advanced_fcb_header-fsrtloplockfsctrl.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2020-12-14:
Adding label.
Thanks @rodwiddowson. I love RFC language. This time of year [RFC968](https://www.rfc-editor.org/rfc/rfc968.txt) might apply as well. ;-)

lorihollasch, 2021-01-11:
@rodwiddowson , thanks as always for your thoughtful issues, and for your suggested language. I consulted with Christian and added an Oplock Synchronization page with a link on all oplock-establishing FsRtl/Flt pages. Christian confirmed that both situations that you identify are MUSTs. (He also confirmed that he missed that he had mistyped "can" vs "cant" on the old OSR post.) Send email if you think anything else is missing. Thanks!

closed 2021-01-11 by lorihollasch