MicrosoftDocs/windows-driver-docs-ddi #1022: Doc page for DEVICE_RESET_INTERFACE_STANDARD DeviceReset function doesn't indicate allowed IRQL ranges

peterwie, 2021-01-08:
This document page doesn't tell me whether I can make the call to issue an FLR or a platform level reset at dispatch level.  This should have allowed caller IRQL ranges.

Similarly the FUNCTION_LEVEL_DEVICE_RESET_PARAMETERS structure (wdm.h) page doesn't indicate the IRQL at which the optional callback can be invoked.
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f72ae897-a27f-1fed-e504-84e4afd7db9b
* Version Independent ID: e824e8fb-bf5b-1bd9-fcab-f09253297487
* Content: [PDEVICE_RESET_HANDLER (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nc-wdm-pdevice_reset_handler)
* Content Source: [wdk-ddi-src/content/wdm/nc-wdm-pdevice_reset_handler.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nc-wdm-pdevice_reset_handler.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-01-21:
Thanks Peter, I just pushed the update - should be on the live site in ~30 min

closed 2021-01-21 by tedhudek

peterwie, 2021-01-21:
Cool.  Thanks Ted.

-p