MicrosoftDocs/windows-driver-docs-ddi #1025: DXGKDDI_RENDER age is not rendering properly

shaboinkin, 2021-01-13:
See screenshot. Happens on Safari and Firefox. 

[image]

This makes it hard to follow the information on the page.
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 35d49678-517e-cbe3-0020-5a58cb2e864d
* Version Independent ID: 1c91d8e4-f061-8cd5-eabb-3009ee4fd39c
* Content: [DXGKDDI_RENDER (d3dkmddi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmddi/nc-d3dkmddi-dxgkddi_render)
* Content Source: [wdk-ddi-src/content/d3dkmddi/nc-d3dkmddi-dxgkddi_render.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dkmddi/nc-d3dkmddi-dxgkddi_render.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2021-01-13:
Thanks for the feedback, @shaboinkin 

Repro on Chrome and Edge, too, @lorihollasch .

lorihollasch, 2021-01-16:
Thanks for reporting this, @shaboinkin  - I've fixed the formatting issue and all is good now.

closed 2021-01-16 by lorihollasch