MicrosoftDocs/windows-driver-docs-ddi #1032: EvtDeviceEnableWakeAtBus and EvtDeviceDisableWakeAtBus have no reference back to the function used to set them

rahul-ramadas, 2021-01-23:
[EvtDeviceEnableWakeAtBus](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfpdo/nc-wdfpdo-evt_wdf_device_enable_wake_at_bus) and [EvtDeviceDisableWakeAtBus](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfpdo/nc-wdfpdo-evt_wdf_device_disable_wake_at_bus) need to have a reference back to [WdfPdoInitSetEventCallbacks](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfpdo/nf-wdfpdo-wdfpdoinitseteventcallbacks), the function used to set them. It was hard to discover this information. Other callbacks set by the same function, like [EvtDeviceReportedMissing](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfpdo/nc-wdfpdo-evt_wdf_device_reported_missing) do have that.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4df2c1a5-d5a1-b6af-9ece-8964d8cdca17
* Version Independent ID: 61208f2d-3314-1cb7-4d75-8570cbebc5a4
* Content: [EVT_WDF_DEVICE_ENABLE_WAKE_AT_BUS (wdfpdo.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfpdo/nc-wdfpdo-evt_wdf_device_enable_wake_at_bus)
* Content Source: [wdk-ddi-src/content/wdfpdo/nc-wdfpdo-evt_wdf_device_enable_wake_at_bus.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfpdo/nc-wdfpdo-evt_wdf_device_enable_wake_at_bus.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-01-23:
Thanks for taking the time to let us know, Rahul - I'll get this done soon.

tedhudek, 2021-01-26:
Just pushed the fix, waiting for update to live site

tedhudek, 2021-01-26:
Live site has the fixes.  Thanks again for letting us know, Rahul.

closed 2021-01-26 by tedhudek