MicrosoftDocs/windows-driver-docs-ddi #1038: Describe Memory Lifetimes of `FWPS_CALLOUT_CLASSIFY_FN` arguments for WFP Callouts

DepthDeluxe, 2021-01-31:
While building a callout driver that hooks into the ALE layer that require caching values from the `inFixedValues` and `inMetaValues` arguments.  Specifically, functions like [RtlCopySid](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-rtlcopysid) and [RtlUnicodeStringToAnsiString](https://docs.microsoft.com/en-us/windows/win32/api/winternl/nf-winternl-rtlunicodestringtoansistring) require calling from a passive IRQL context which is not possible inside the WFP callout directly.

Based on the memory lifetime of these callout arguments, I see one of two ways of solving this problem:

(1) If the lifetime is longer than just the individual trigger at the ALE layer, I could copy the pointers into my buffer and cache for subsequent usage later on.

(2) If the lifetime is coupled to the individual trigger at ALE layer, I could instead pend the ALE packet using [FwpsPendOperation0](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nf-fwpsk-fwpspendoperation0), send the packet off to a worker thread in the passive context, call the PASSIVE IRQL context functions to copy the data into buffers owned by my callout driver, and then complete the packet operation.

Documenting this lifetime information would be very helpful in making the above types of design decisions.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0a478874-0fb5-c185-81ac-826e52749e41
* Version Independent ID: d4dbd7f7-819d-674c-24ca-1b4c02c413eb
* Content: [FWPS_CALLOUT_CLASSIFY_FN0 (fwpsk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nc-fwpsk-fwps_callout_classify_fn0)
* Content Source: [wdk-ddi-src/content/fwpsk/nc-fwpsk-fwps_callout_classify_fn0.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fwpsk/nc-fwpsk-fwps_callout_classify_fn0.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

DepthDeluxe, 2021-01-31:
I see that the [FwpsPendOperation0](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nf-fwpsk-fwpspendoperation0) function explicitly describes the case I am encountering related to needing to evaluate in the PASSIVE IRQL context.  In that case, my understanding is that I would have to go with option (2).

Perhaps it would be good to add a bit of text in the remarks section that if you need to access any of these function arguments in the passive context and you are in a layer that supports pending packets, you can pend them for subsequent processing.

aviviano, 2021-03-25:
Hi @DepthDeluxe, thanks for opening this issue. We've reviewed your comments and have decided to leave the page as-is for now as the case in question is already called out in the FwpsPendOperation documentation. Thanks again.

closed 2021-03-25 by aviviano