MicrosoftDocs/windows-driver-docs-ddi #1039: Example needed for KeSetHardwareCounterConfiguration

ikhalil, 2021-02-04:
This page need either more details or at least some code example on how to use this API. Currently using any index value in the hardware counter structure always return success, making it hard to understand how to use the API.

Additionally, the Hardware Counter struct documentation related to this need a better explanation on how to find the index of the counters we are interested in.

Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c27dbd8c-d6b0-8259-5093-bf3a7651b546
* Version Independent ID: 7aea9698-0ece-2126-dee5-dfbaa402ab7a
* Content: [KeSetHardwareCounterConfiguration function (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-kesethardwarecounterconfiguration)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-kesethardwarecounterconfiguration.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-kesethardwarecounterconfiguration.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

ikhalil, 2021-02-04:
To add more details, I have been trying to figure out how to use this API but I cannot seem to figure out the correct index value needed for the hardware counter (this documentation is the only information available online about this API it seems)

tedhudek, 2021-02-05:
I don't have much experience here, but it looks like this ties into hardware counter profiling, which is described at https://docs.microsoft.com/en-us/previous-versions/windows/desktop/hcp/hcp-reference.  Can you use the Threadprofiling APIs to get counter info?  For example -- https://docs.microsoft.com/en-us/windows/win32/api/winbase/nf-winbase-querythreadprofiling.  I added both links to the ref page (should be live soon).

I checked the driver samples repo but did not see any examples of KeSetHardwareCounterConfiguration.

You could also try the forums, if this is still an issue:

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Thanks for reaching out and good luck!

closed 2021-02-05 by tedhudek

cmuratori, 2022-08-15:
This does not actually fix the documentation problem described by the OP. Calling QueryThreadProfiling simply tells you if you have previously called EnableThreadProfiling on a thread. It does not help you figure out how to call KeSetHardwareCounterConfiguration.

KeSetHardwareCounterConfiguration is _supposed_ to control which hardware counters are enabled for profiling. According to the documentation, you can _supposedly_ call it to enable hardware performance counters that will then be returned by future calls to ReadThreadProfilingData,

However, nobody (to the best of my knowledge) has ever figured out how to get this to work, because there is no documentaiton from Microsoft about how to correctly call KeSetHardwareCounterConfiguration. You can call it all you want - and if you pass a properly formatted array and correct count, it will even return ERROR_SUCCESS - but you will still not get any counters coming out of ReadThreadProfilingData.

So this issue should be reopened. Somebody needs to publish a working example of KeSetHardwareCounterConfiguration so people outside Microsoft can actually use this API. As it currently stands, the KeSetHardwareCounterConfiguration/EnableThreadProfiling/ReadThreadProfilingData documentation clearly states that you can enable and retrieve performance counters, but lack sufficient explanation for anyone to figure out how to make that actually happen.

\- Casey