MicrosoftDocs/windows-driver-docs-ddi #1043: Documentation for parameters is incomplete

darugeri, 2021-02-09:
The documentation for `Maxmin` and `HighWater` parameters is missing.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3b5de4af-42fe-292a-7601-8f23230940ac
* Version Independent ID: e392c09c-cf17-77b1-c8e3-a14bd0548ba8
* Content: [IoInitializeRemoveLock macro (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-ioinitializeremovelock#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-ioinitializeremovelock.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-ioinitializeremovelock.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

darugeri, 2021-02-09:
Also, the code snippet does not have parameter types nor SAL annotations (if applicable).

darugeri, 2021-02-09:
> Also, the code snippet does not have parameter types nor SAL annotations (if applicable).

I just realized is a macro, so I don't know if it should have it in the first place. If not, then please disregard my comment.

tedhudek, 2021-02-10:
Thanks again for the sharp eye, Daniel!  Turns out we had spurious whitespace in the token markers for those param description - I just pushed the fix, ETA 30 min to hit live.

We can override the autogen syntax block but I'm inclined to leave as-is since as you say, it's a macro.  Let me know if you feel differently and we can fake one up.

closed 2021-02-10 by tedhudek