MicrosoftDocs/windows-driver-docs-ddi #1055: Versions for each file info class

rodwiddowson, 2021-03-11:
Hey Lori,

It would be really handy if we could get an indication of which build number included which `FileInfoClass`.

I may be wrong but in my test against 19041 `FileNetworkOpenInformation` returns `STATUS_INVALID_PARAMETER` (which is a pain since I just added support for that to my filter).

I know that Stat and StatLx have worked since forever and I have seen a certain AV product using `FileCaseSensitiveInformation` so I know that works.

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 226fe146-f4fb-0df5-4f5b-c7e577b4e6d1
* Version Independent ID: d34f936a-8045-534c-bdde-fa80f65cd581
* Content: [NtQueryInformationByName function - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ntqueryinformationbyname)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ntqueryinformationbyname.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntqueryinformationbyname.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2021-03-13:
Arg @rodwiddowson , sorry to say but FileNetworkOpenInformation is not supported at all - probably a cut and paste error! I've removed it from the page, and have added version info for the remaining 3 supported FileInfoClasses. Changes will go live soon (commit is 97a44c32)

closed 2021-03-13 by lorihollasch

rodwiddowson, 2021-03-13:
😁