MicrosoftDocs/windows-driver-docs-ddi #1056: POOL_TYPE isn't a standard enum as shown

shicklinFES, 2021-03-15:
The enum implies that each entry increments by 1 from 0 which is not the case.  e.g. from this document PagedPool would now be `2` while it has historically been `1` and in reality it still is, as the entry in wdm.h makes `NonPagedPoolExecute=NonPagedPool`.  Later entries are also incorrect when relying on the assumption of standard enum.  I appreciate if you are only targeting one OS or always using the latest and greatest development kit then this issue may not be as much of a concern.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9654d4be-9f3d-444a-97df-0c2563c4adc7
* Version Independent ID: 98b7ca6b-0067-82be-98f0-c8d96148e6fd
* Content: [_POOL_TYPE (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_pool_type)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_pool_type.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_pool_type.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-03-16:
Hi @shicklinFES thanks for the feedback, and good point.  Do you have a desired outcome, i.e. how would you want the enum to display in the syntax section?  These get auto-generated but it is possible to override manually.

shicklinFES, 2021-03-16:
> [EXTERNAL EMAIL] DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

Hi,
Thanks for the quick reply.  It would be nice if the enum showed what the value was where it deviates from what would normally be expected.  Either the value or referencing other enum values would be fine. I think doing this would clear up any ambiguity.

Thanks

Steve

tedhudek, 2021-03-16:
Hi Steve - I'm not actually sure how to do that.  I did compare the enum between Vb and current Fe builds in WIP, and it's the same.  How would you generate such a markup?  Do you have your own mockup of it that you would like to propose for the syntax block on the doc page?

shicklinFES, 2021-03-17:
A rough suggestion might be the following.  I've taken the entry from wdm.h and removed comments, SAL and where the values are set but I believe the ordering of the enum would give that value anyway (might be worth a double check).

Hope this helps

```
typedef enum _POOL_TYPE {
    NonPagedPool,
    NonPagedPoolExecute = NonPagedPool,
    PagedPool,
    NonPagedPoolMustSucceed,
    DontUseThisType,
    NonPagedPoolCacheAligned,
    PagedPoolCacheAligned,
    NonPagedPoolCacheAlignedMustS,
    MaxPoolType,
    NonPagedPoolBase = 0,
    NonPagedPoolBaseMustSucceed = NonPagedPoolBase + 2,
    NonPagedPoolBaseCacheAligned = NonPagedPoolBase + 4,
    NonPagedPoolBaseCacheAlignedMustS = NonPagedPoolBase + 6,
    NonPagedPoolSession = 32,
    PagedPoolSession = NonPagedPoolSession + 1,
    NonPagedPoolMustSucceedSession = PagedPoolSession + 1,
    DontUseThisTypeSession = NonPagedPoolMustSucceedSession + 1,
    NonPagedPoolCacheAlignedSession = DontUseThisTypeSession + 1,
    PagedPoolCacheAlignedSession = NonPagedPoolCacheAlignedSession + 1,
    NonPagedPoolCacheAlignedMustSSession = PagedPoolCacheAlignedSession + 1,
    NonPagedPoolNx = 512,
    NonPagedPoolNxCacheAligned = NonPagedPoolNx + 4,
    NonPagedPoolSessionNx = NonPagedPoolNx + 32,
} POOL_TYPE;
```

tedhudek, 2021-03-29:
Just pushed the change - should be on the live site in 30 min or so.

Thanks again for the suggestion!

closed 2021-03-29 by tedhudek