MicrosoftDocs/windows-driver-docs-ddi #1062: No webpage to intro the structure of Info.GenErrDescriptorV2

katyhuang72727, 2021-03-26:
It should be supported on current shipping OS, not sure why there's no intro of this structure on website.
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 52ffd6d4-4487-5390-6976-706eabd8d239
* Version Independent ID: c2a04de3-0b74-d895-677a-0c87cca300cd
* Content: [_WHEA_ERROR_SOURCE_DESCRIPTOR (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/ns-ntddk-_whea_error_source_descriptor)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_whea_error_source_descriptor.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_whea_error_source_descriptor.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-03-29:
Very good point @katyhuang72727, thanks for letting us know.  You are right - I will add a structure doc page for `WHEA_GENERIC_ERROR_DESCRIPTOR_V2`.  In the meantime, look in ntddk.h and you can find thorough code comments for that struct (line 17522).

I'll keep this issue open until the new page is up.

tedhudek, 2021-03-30:
The new page is up!  We're still working on descriptions for the last 3 members, but you can see everything else at https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/ns-ntddk-whea_generic_error_descriptor_v2.

Thanks again for the feedback.

closed 2021-03-30 by tedhudek