MicrosoftDocs/windows-driver-docs-ddi #1066: IoGetDevicePropertyData - STATUS_OBJECT_NAME_NOT_FOUND on Windows 7

sormo, 2021-04-07:
It seems that function `IoGetDevicePropertyData` returns `STATUS_OBJECT_NAME_NOT_FOUND` on Windows 7. For example for `DEVPKEY_Device_ClassGuid`.   

I've found mention about this on osr and stackoverflow:  
https://community.osr.com/discussion/240164/iogetdevicepropertydata-returns-status-object-name-not-found  
https://stackoverflow.com/questions/15353557/iogetdevicepropertydata-returns-status-object-name-not-found  

I don't see information about this in documentation for `IoGetDevicePropertyData` where is stated that this function just works from Vista. If this function can't get same properties which are possible to get in user mode it would be good to mention this.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2a9c40b7-a0c4-f5fe-433b-d3a547e5b3d2
* Version Independent ID: 36d428eb-8dc2-5c18-3539-7af342deaee6
* Content: [IoGetDevicePropertyData function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-iogetdevicepropertydata)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iogetdevicepropertydata.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iogetdevicepropertydata.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-04-23:
Hi @sormo, thanks for the feedback.  I'm looking into this, but in the short term, would you want to do something like https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-iogetdeviceproperty#devicepropertyclassguid for pre-Win8?

tedhudek, 2021-05-01:
Pre-Win8 I think all properties could be accessed from user mode but only a subset of them could be accessed from kernel mode. Starting in Win8 the k-m access was broadened to match the u-m access.

Identifying and updating the docs for the subset unfortunately falls below the priority line for our writers at the moment.  If you want to open a PR with details on your experience, happy to review it.

Thanks again for the feedback!

closed 2021-05-01 by tedhudek