MicrosoftDocs/windows-driver-docs-ddi #1068: Confusing/misleading documentation for vital parameter

mr-intj, 2021-04-13:
After puzzling why the data being written by my driver to the output buffer of an IOCTL request wasn't showing up on the client side (`DeviceIoControl`), I came across a suggestion on stackoverflow.com that it's _necessary_ to call `WdfRequestCompleteWithInformation` (vs `WdfRequestComplete`) and pass back the number of bytes written to the output buffer for any I/O requests that have output parameters.

The rest of this issue is based upon that assumption...

The summary for this API reads:

> The WdfRequestCompleteWithInformation method stores completion information and then completes a specified I/O request with a supplied completion status.

The major difference between this summary and the summary for WdfRequestComplete is _"stores completion information"_. It doesn't make clear that IOCTL requests _that write information to an output buffer_ **must** use WdfRequestCompleteWithInformation to complete a request.

The documentation for the **Information** parameter says:

> [in] Driver-defined completion status information for the request, such as the number of bytes that were transferred.

"Driver-defined" here seems misleading (as though the meaning of the information passed through this parameter is up to the driver developer). That doesn't seem to be the case. The same can be said for "such as ..." which suggests (to me, at least) that this API provides some kind of out-of-band return path for driver-specific data.

Similarly, the documentation page for WdfRequestCompleted only mentions WdfRequestCompleteWithInformation in passing:

> Instead of calling WdfRequestComplete, the driver can call WdfRequestCompleteWithInformation or WdfRequestCompleteWithPriorityBoost.

...and it doesn't explain why you'd use one vs. the other.

I'd suggest explicitly calling out the fact that any I/O requests that write to the output buffer _must_ call WdfRequestCompleteWithInformation to complete the request. I'd make this clear on the doc pages for _both_ APIs.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 89bc2f40-06fb-7161-7d2b-d3eee8023b59
* Version Independent ID: a452150b-448e-c1f1-a3dd-d66f009cb233
* Content: [WdfRequestCompleteWithInformation function (wdfrequest.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfrequest/nf-wdfrequest-wdfrequestcompletewithinformation)
* Content Source: [wdk-ddi-src/content/wdfrequest/nf-wdfrequest-wdfrequestcompletewithinformation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfrequest/nf-wdfrequest-wdfrequestcompletewithinformation.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-04-24:
Thanks for the feedback @mr-intj, great points.  I'm investigating but in the meantime here's a bit of related info --

https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/completing-i-o-requests#-providing-completion-information

Last bullet of https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/supporting-kernel-mode-clients-in-umdf-drivers#restrictions-on-kernel-mode-drivers

mr-intj, 2021-04-25:
Thanks Ted. From the links you sent, it looks like there is a capability to send additional out-of-band(?) information:

> ...such as handles to memory objects that represent the request's buffers, or bus-specific information

...so it appears that the docs aren't _wrong_ so much as vague enough to be potentially misleading.

Thanks for the info, and for looking into this.

---
Scott Smith

tedhudek, 2021-05-01:
Thanks again for the feedback, Scott - I updated the docs in the commit above.

closed 2021-05-01 by tedhudek