MicrosoftDocs/windows-driver-docs-ddi #1078: Incorrect Link

Rohit-MSFT, 2021-04-25:
[Enter feedback here]

The link for "**_SECURITY_SUBJECT_CONTEXT_**" in below line is incorrect. 
[in] Pointer to the **_SECURITY_SUBJECT_CONTEXT_** structure that specifies the subject's captured security context.

it currently points to:
https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/eprocess

it should point to: 
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_security_subject_context

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 27f6f27a-1ccc-2381-dfd2-19cf3a4da0db
* Version Independent ID: 30d125f7-9504-7930-bc25-9727a0955b4e
* Content: [SeAccessCheck function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-seaccesscheck)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-seaccesscheck.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-seaccesscheck.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-04-26:
Thanks Rohit!  Good to see you again :)

mhopkins-msft, 2021-04-26:
Hi @Rohit-MSFT,

Thanks for reporting this. The topic has been updated.

 ~ *Mark*

closed 2021-04-26 by mhopkins-msft