MicrosoftDocs/windows-driver-docs-ddi #1079: Cannot find this file in 19041 WDK

guliteb, 2021-04-26:
Cannot find this file in 19041 WDK

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: aa3bc7ca-8364-0eb6-414a-70d1af05de83
* Version Independent ID: 2248b496-6a59-eaea-fc76-7f14b39e3f4e
* Content: [Acpiioct.h header - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/acpiioct/)
* Content Source: [wdk-ddi-src/content/acpiioct/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/acpiioct/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-04-26:
On my system, with a default WDK install, the file is located here:

C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\shared\acpiioct.h

[image]

Is the acpiioct.h file missing in your WDK installation?

guliteb, 2021-04-27:
Thanks. I re-installed WDK and the file is there. Thanks.

closed 2021-04-27 by guliteb