MicrosoftDocs/windows-driver-docs-ddi #1082: PCI DEVICE

Haleymiranda79, 2021-05-02:
How do you install a missing driver for your PCI  device?

tedhudek, 2021-05-04:
Thanks for reaching out to the Windows Driver Developer Documentation team. We create the documentation for companies that undertake driver development and deployment from a commercial perspective, and we’re staffed to solve issues with the supporting documentation. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation. 

I found this forum discussion that might be relevant - https://forums.tomshardware.com/threads/pci-device-drivers-not-installed.3538918/.  If not, you could try posting your question there.

Good luck!

closed 2021-05-04 by tedhudek

Haleymiranda79, 2021-05-17:
Thanks @tedhudek I just read this.