MicrosoftDocs/windows-driver-docs-ddi #1083: KPRIORITY lack of information

EmmanuelLEPELLETIER, 2021-05-02:
Could you add some information about KPRIORITY parameter ?
Which values can be used, how to set it correctly, the behaviour, ...
Thanks :-)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4259dc47-0e73-3102-d12f-1f337d5bea73
* Version Independent ID: 9d150a84-5d04-bfb9-0237-7021c5e99e41
* Content: [KePulseEvent function (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-kepulseevent)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-kepulseevent.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-kepulseevent.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-05-02:
Thanks for the question, @EmmanuelLEPELLETIER.  I'm investigating but in the meantime here are a couple suggestions.

`Wdm.h` includes:

```
#define LOW_PRIORITY 0              // Lowest thread priority level
#define LOW_REALTIME_PRIORITY 16    // Lowest realtime priority level
#define HIGH_PRIORITY 31            // Highest thread priority level
#define MAXIMUM_PRIORITY 32         // Number of thread priority levels
...
typedef LONG KPRIORITY;
```

Also, these doc pages have a little info on KPRIORITY:

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-kequeryprioritythread ("Thread priorities range from 0 to 31, where 0 is the lowest priority and 31 is the highest.")

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-kesetprioritythread

tedhudek, 2021-05-02:
I also see a couple uses of **KeSetPriorityThread** in https://github.com/microsoft/Windows-driver-samples.

tedhudek, 2021-05-19:
Hi again @EmmanuelLEPELLETIER, apologies for the delay.  I've updated the description of the **Increment** parameter on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-kepulseevent to address your question.  Hope it helps!

Thanks again for the feedback.

closed 2021-05-19 by tedhudek