MicrosoftDocs/windows-driver-docs-ddi #1086: FwpsCompleteOperation will cause system down, please give some examples

robotwxy, 2021-05-05:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7236f453-71c0-24fa-6c7a-273ad73b44d6
* Version Independent ID: 627d461f-7885-9865-378b-962a544c6202
* Content: [FwpsCompleteOperation0 function (fwpsk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nf-fwpsk-fwpscompleteoperation0)
* Content Source: [wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpscompleteoperation0.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpscompleteoperation0.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

robotwxy, 2021-05-05:
FwpsCompleteOperation

aviviano, 2021-05-05:
Hi @robotwxy, which function did you use when you had a system crash, FwpsCompleteOperation0  or FwpsCompleteOperation?

aviviano, 2021-05-11:
Hi @robotwxy, please see the [WFPSampler](https://github.com/Microsoft/Windows-driver-samples/tree/master/network/trans/WFPSampler) for usage examples. Thanks for opening this issue. If you have further questions please open a new issue.