MicrosoftDocs/windows-driver-docs-ddi #1089: MmEnumerateSystemImagesEx doesn't exist

Alan-Jowett, 2021-05-10:
MmEnumerateSystemImagesEx doesn't exist.

This API doesn't exist in the listed version of the Windows kernel.

Repro steps:
```
    RtlInitUnicodeString(&unicode_string, L"MmGetSystemRoutineAddressEx");
    *(PVOID*)&mm_get_system_routine_address_ex = MmGetSystemRoutineAddress(&unicode_string);
```

Expected result:
MmGetSystemRoutineAddressEx should be found.

Actual results:
MmGetSystemRoutineAddressEx is not present.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4873b0f3-f061-4fa1-de0c-ebb3b626dabe
* Version Independent ID: 9783a8d3-b25a-bbfa-4a2e-19739c8afd78
* Content: [MmGetSystemRoutineAddressEx function - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-mmgetsystemroutineaddressex)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-mmgetsystemroutineaddressex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-mmgetsystemroutineaddressex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-05-13:
Hi @Alan-Jowett, thanks for the feedback.

(In your first sentence above you say MmEnumerateSystemImagesEx, but I'm pretty sure you mean MmGetSystemRoutineAddressEx since you use that in other references.)

I'm thinking you might be seeing this behavior due to the conditional that appears in wdm.h:

```
#if defined(XBOX_SYSTEMOS) || defined(XBOX_GAMECOREOS)

_IRQL_requires_max_(PASSIVE_LEVEL)
NTKERNELAPI
PVOID
NTAPI
MmGetSystemRoutineAddressEx (
    _In_ PUNICODE_STRING ModuleName,
    _In_ PSTR FunctionName
    );

#endif // defined(XBOX_SYSTEMOS) || defined(XBOX_GAMECOREOS)
```

Alan-Jowett, 2021-05-13:
Thanks, I fixed the title. My issue is that the MSDN doc appears to be wrong as it states minimum version is Windows 10, version 2004, which is clearly not true (kernel doesn't appear to have the symbol at all). 

[image]

tedhudek, 2021-05-19:
Just pushed a doc update to change the requirements listing to Xbox.  Should appear on the live site within 30 min.

Thanks for the head-up @Alan-Jowett!

closed 2021-05-19 by tedhudek