MicrosoftDocs/windows-driver-docs-ddi #1106: Undocumented behaviour of ZwUnloadDriver

gsuberland, 2021-05-30:
The ZwUnloadDriver documentation is missing information about the behaviour of the function in certain situations.

Specifically:

1. Can ZwUnloadDriver unload a driver whose `DRIVER_OBJECT` descriptor has no unload callback set?
2. If ZwUnloadDriver cannot be used to unload a driver without an unload callback, what error code is returned?
3. Can a driver call ZwUnloadDriver on itself?
4. Can a driver call ZwUnloadDriver on itself if it does not have an unload callback set?
5. If a driver cannot call ZwUnloadDriver on itself, what error code is returned?
6. If a driver can call ZwUnloadDriver on itself, but there are hazards in doing so, what are those hazard? What precautions should developers take to avoid them?
7. Is there a supported/preferred way for a driver to unload/stop itself?
8. Does the answer to any of these questions change depending on whether the driver is a non-WDM, WDM Bus, WDM Function, or WDM Filter driver? (the last one is partially answered already with the warning on the page)

It would also be helpful to document how the SCM behaves in regard to drivers changing the DriverUnload field of the `DRIVER_OBJECT` descriptor at runtime. It appears that the `sc` command (and by extension the SCM) queries the DriverUnload field of the driver object in order to apply the Stoppable flag on driver services - i.e. a device service is only marked as stoppable if the DriverUnload callback is non-null. But what isn't clear is whether this is checked on every usermode query, only at the point of device creation, or if there is some more complex abstraction or caching in between. I'm not sure whether this is something that would be documented on the ZwUnloadDriver page specifically, but it should probably link there since it's relevant.

Some of these questions come up in situations where a driver service should not be stoppable in normal operation, and thus should not be seen as stoppable by the SCM, but may need to be unloaded in certain application-specific circumstances.

I had a bit of a dig around for the answer to these questions, but was mostly met with OSR forum posts and StackOverflow questions where the answers were vague, conflicting, guesses, or missing entirely.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6cbf4254-1a19-e303-a6fa-534fa0c7d870
* Version Independent ID: 293e90ad-8e2f-ac82-3384-70e64f38ff80
* Content: [ZwUnloadDriver function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-zwunloaddriver)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-zwunloaddriver.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-zwunloaddriver.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2021-07-29:
Hi @gsuberland, apologies for taking so long to get back to you.

I just updated https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-zwunloaddriver with some additional guidance that will hopefully clarify some of your questions.  The changes are in commit e6e4203a3ec32f83567752bfa5357d3cd92701b1 if you want to have a look.

Hope this helps -- and thanks for caring about the docs!

closed 2021-07-29 by tedhudek

gsuberland, 2021-08-03:
While I appreciate the clarification in regard to the error code when `DriverUnload` is not set, the documentation changes didn't answer some of my more important questions:

I presume from "it is not recommended that a driver call ZwUnloadDriver on itself" that there are indeed hazards in doing so. What kind of hazards are most applicable here? What precautions should developers take to avoid them if they do need to self-unload? Better yet, is there a recommended way for a driver to unload/stop itself?