MicrosoftDocs/windows-driver-docs-ddi #1111: Bug in the sample code

emild, 2021-06-25:
[Enter feedback here]
There's a small bug in the sample code: the assignment to status should occur be before the break statements

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7b38d318-86eb-e6e6-c981-9b258f8c340c
* Version Independent ID: 087dcbd4-90ca-8203-d735-f8f724d767e4
* Content: [WdfRequestSetInformation function (wdfrequest.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfrequest/nf-wdfrequest-wdfrequestsetinformation)
* Content Source: [wdk-ddi-src/content/wdfrequest/nf-wdfrequest-wdfrequestsetinformation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfrequest/nf-wdfrequest-wdfrequestsetinformation.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-07-01:
Hi @emild, we appreciate the report, thank you.  Would you have any interest in opening a PR with your proposed fix?

emild, 2021-07-04:
Hi Ted,

I created a PR

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/pull/1117

                                         Regards,

                                                              Emil

tedhudek, 2021-07-06:
Hi Emil, thank you so much!  The PR looked great.  I just completed it - payload should reach the live site within 5-10 min.

Thanks again for caring about the docs :)

closed 2021-07-06 by tedhudek