MicrosoftDocs/windows-driver-docs-ddi #1121: Documentation for KeRundownQueue seems outdated wrt. currently waiting threads

MicheleCicciottiWork, 2021-07-15:
The documentation for `KeRundownQueue` seems outdated. [In the remarks, it says](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-kerundownqueue.md#-remarks):

> <b>KeRundownQueue</b> should never be called for a queue if any threads are waiting on the queue object.

This seems to be true up until Windows XP: on those platforms, `KeRundownQueue` doesn't appear to satisfy existing waits, potentially leaving threads waiting forever. However, at least since Windows 7, `KeRundownQueue` causes any calls to `KeRemoveQueue` to fail with `STATUS_ABANDONED`, [as correctly documented in the page for `KeRemoveQueue`](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-keremovequeue.md#-returns).

tedhudek, 2021-07-15:
Hi @lorihollasch, assigning to you since this issue references https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-kerundownqueue.

lorihollasch, 2021-08-31:
@MicheleCicciottiWork , great catch and you are absolutely right about the behavior change starting in Windows 7. I have updated the page accordingly. Thanks for filing an issue and for the additional useful info RE OS versions. Apologies for my slow response!

closed 2021-08-31 by lorihollasch