MicrosoftDocs/windows-driver-docs-ddi #1134: The Link to POOL_FLAGS is not working....

pankajbharti, 2021-07-21:
The Link to POOL_FLAGS is not working. The link takes me to : https://review.docs.microsoft.com/en-us/windows-hardware/drivers/kernel/pool_flags?branch=kernel-manganese which is not valid
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cb4545d5-fc4e-4a0d-1634-c9d059814127
* Version Independent ID: 57255a21-1001-2a98-5bc6-19869d48fa0b
* Content: [ExAllocatePool3 - Windows drivers](https://review.docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exallocatepool3?branch=kernel-manganese)
* Content Source: [https://cpubwin.visualstudio.com/DefaultCollection/drivers/_git/wdk-ddi/?path=%2fwdk-ddi-src%2fcontent%2fwdm%2fnf-wdm-exallocatepool3.md&version=GBkernel-manganese&_a=contents](https://cpubwin.visualstudio.com/DefaultCollection/drivers/_git/wdk-ddi/?path=%2fwdk-ddi-src%2fcontent%2fwdm%2fnf-wdm-exallocatepool3.md&version=GBkernel-manganese&_a=contents)
* Product: **windows-hardware**
* Technology: **windows-devices**
* Git Login: **tedhudek**
* Microsoft Alias: **tedhudek**

tedhudek, 2021-07-22:
Hi @pankajbharti, it seems that somehow you got on the review site. (See the Content link in the Document Details section above -- this is the page that you were on when you opened the issue, and it's a page on the review site.)  The link works normally when you click it from the live page, which is https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exallocatepool3.

Do you know how you reached the review page?  

Thanks for the report.

tedhudek, 2021-07-23:
Hi Pankaj, I'm going to close this one out since there is no action item.  If you know how you reached the review page, please let me know and perhaps there is a bug in that nav path that we can fix.

Enjoy your summer!

closed 2021-07-23 by tedhudek