MicrosoftDocs/windows-driver-docs-ddi #1169: Is WDI_P2P_DISCOVER_TYPE_SCAN_ONLY actually the Scan Phase in  the Wi-Fi Direct specification?

yujuiwu, 2021-08-19:
[Enter feedback here]

Hello, 

I'm a bit confused about the definition:
> WDI_P2P_DISCOVER_TYPE_SCAN_ONLY
Device discovery occurs only during the scan phase. Adapter should scan each channel at least once every 250 milliseconds.

Is this the actually the "Scan Phase" in the Wi-Fi Direct specification? 

If we have 11 channels (1 to 11), does this mean that the adapter should scan each of these 11 channels at least once every 250 milliseconds?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 89e0102b-75d1-4a22-96db-e09ede4d0c16
* Version Independent ID: 34905805-0c5c-c87c-81f9-26f610119cac
* Content: [_WDI_P2P_DISCOVER_TYPE (wditypes.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wditypes/ne-wditypes-_wdi_p2p_discover_type)
* Content Source: [wdk-ddi-src/content/wditypes/ne-wditypes-_wdi_p2p_discover_type.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wditypes/ne-wditypes-_wdi_p2p_discover_type.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2021-08-19:
Hi @yujuiwu. Thanks for your questions. Yes to both. The _SCAN_ONLY and _FIND_ONLY values in the enumeration are references to the procedures described in sections 3.1.2.1.2 and 3.1.2.1.3 of the Wi-Fi Alliance’s Wi-Fi Direct specification.

closed 2021-08-19 by aviviano

yujuiwu, 2021-08-20:
> Hi @yujuiwu. Thanks for your questions. Yes to both. The _SCAN_ONLY and _FIND_ONLY values in the enumeration are references to the procedures described in sections 3.1.2.1.2 and 3.1.2.1.3 of the Wi-Fi Alliance’s Wi-Fi Direct specification.

Hi @aviviano, 

Thanks for your reply. 

But if we have 11 channels to scan and the adapter should scan each of these channels at least once every 250 milliseconds, each channel could only be allocated around 23 milliseconds. That way only active scan is feasible and we are not considering 5G channels.

May I confirm with you whether this 250 milliseconds is a duration (the amount of time stay on the channel) or a period (time between each scan of the channel)?