MicrosoftDocs/windows-driver-docs-ddi #1185: Resource strings referenced by COUNTED_REASON_CONTEXT (wdm.h)

KalleOlaviNiemitalo, 2021-09-18:
There are two articles about COUNTED_REASON_CONTEXT in this repo:

* [COUNTED_REASON_CONTEXT structure (ntpoapi.h)](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntpoapi/ns-ntpoapi-_counted_reason_context) from [wdk-ddi-src/content/ntpoapi/ns-ntpoapi-_counted_reason_context.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntpoapi/ns-ntpoapi-_counted_reason_context.md)
* [COUNTED_REASON_CONTEXT structure (wdm.h)](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_counted_reason_context) from [wdk-ddi-src/content/wdm/ns-wdm-_counted_reason_context.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_counted_reason_context.md)

<https://github.com/MicrosoftDocs/windows-driver-docs-ddi/issues/1000> was fixed in the ntpoapi.h copy of the documentation, but the wdm.h copy still has the incorrect information about consecutive resource IDs.

I suppose there are three ways to fix this:

- Apply the same changes to the wdm.h copy.
- Move the shared text to a separate file and [`!INCLUDE`](<https://docs.microsoft.com/en-us/contribute/markdown-reference#included-markdown-files> "Included Markdown files - Markdown reference for docs.microsoft.com - Contributor Guide | Microsoft Docs") it in both articles. This would reduce the risk that the articles diverge again in the future.
- Make one of the articles redirect to the other, so that Web search engines only list one article rather than two. This would make it clear to readers where to find the correct documentation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 18ed4424-8459-a3aa-07f9-11b6c498400b
* Version Independent ID: 2b474d49-5980-638b-094c-1693a8f9bd7b
* Content: [_COUNTED_REASON_CONTEXT (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_counted_reason_context)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_counted_reason_context.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_counted_reason_context.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

KalleOlaviNiemitalo, 2021-09-18:
@tedhudek, you might be interested in this because you fixed <https://github.com/MicrosoftDocs/windows-driver-docs-ddi/issues/1000>.

tedhudek, 2021-09-20:
Thanks for filing this, @KalleOlaviNiemitalo.  The fix is building and should appear on the live site within half an hour.

closed 2021-09-20 by tedhudek