MicrosoftDocs/windows-driver-docs-ddi #1189: FileKnownFolderInformation

maharmstone, 2021-09-21:
FileKnownFolderInformation is missing from this list.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a8119f8b-9c83-60f2-ac36-0478b88c4ab0
* Version Independent ID: 404ee010-b7dc-6a9e-7bbe-7cee72809333
* Content: [_FILE_INFORMATION_CLASS (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_file_information_class)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2021-09-22:
Hi @maharmstone,

Thanks so much for bringing this to our attention.  I've added the missing enum value to https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_file_information_class.

Adding it helped me realize that we do not yet have a reference page for the FILE_KNOWN_FOLDER_INFORMATION structure, which does appear in `ntifs.h` in the Windows 11 WDK.  So that's a known issue that we are tracking.

Thanks again for taking the time to report this!

closed 2021-09-22 by tedhudek

maharmstone, 2021-09-22:
Cheers!

tedhudek, 2021-09-29:
BTW @maharmstone, thanks to @lorihollasch we now have a ref page for the FILE_KNOWN_FOLDER_INFORMATION structure --

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-file_known_folder_information

maharmstone, 2021-09-29:
Thanks both!