MicrosoftDocs/windows-driver-docs-ddi #1190: Wrong setup macro used for AePushLock field

cgallred, 2021-09-28:
The AePushLock field, added in Win 10 Manganese, has to be initialized using FsRtlSetupAdvancedHeaderEx**2** (current doc says "FsRtlSetupAdvancedHeaderEx"). 

Also need to create a page for FsRtlSetupAdvancedHeaderEx2 so this page can link to it.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cb2d22bd-f5a3-890a-6703-1b2a1d7047d3
* Version Independent ID: cebba538-54f4-41e5-b7ec-6e0e724cfac5
* Content: [FSRTL_ADVANCED_FCB_HEADER (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header?source=docs)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2021-09-29:
Nothing to do with the documentation, but wouldn't it be better to fix this problem once and for all by having a variant which take the length of the structure ?  Then we won't be asking ourselves whether the next function will be `FooEx3` or `FooEx2Ex` :-)

With that done you make the whole thing portable by hiding the length in a macro ```#define foo(ptr) fooEx3(ptr, sizeof(*ptr))```

lorihollasch, 2021-09-30:
@cgallred , thanks for reporting this! I've corrected this page, added a section to define auto-expand push locks, and wrote Ex2 page. Also added missing Allocate/FreeAePushLock pages.

closed 2021-09-30 by lorihollasch

cgallred, 2021-09-30:
@rodwiddowson I don't quite understand your comment. Each succeeding macro for initializing FSRTL_ADVANCED_FCB_HEADER has to take new parameters to initialize fields of the structure that didn't exist previously. What would be accomplished by taking the size?

Side note, we adopted naming conventions a few years ago that eliminate the "Ex" suffix. Nowadays new API versions get a numeric suffix (c.f. CreateFile, CreateFile2). Existing Ex-suffixed APIs are relics of the old ways. :-)

rodwiddowson, 2021-09-30:
@cgallred 
>  Each succeeding macro for initializing FSRTL_ADVANCED_FCB_HEADER has to take new 
> parameters to initialize fields of the structure that didn't exist previously. What would be 
> accomplished by taking the size?

O.I.C the problem - I have to provide the address of the mutex and the pushlock and the location of the file context pointer.  I could see that you could have fixed the first two by embedding the structure then you don't get to let one lock serve multiple streams.  The last absolutely has to be externally provided.

Thanks (& sorry)