MicrosoftDocs/windows-driver-docs-ddi #1214: NDIS_OFFLOAD_PARAMETERS_UDP_SEG_DISABLED is not defined in WDK

sandilyabhagi, 2021-11-10:
NDIS_OFFLOAD_PARAMETERS_UDP_SEG_DISABLED is not defined in WDK. We are using WDK 22000.1

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 76b8c5f9-9bf8-3960-7a29-e4a81c7b661a
* Version Independent ID: 649b3a4c-1c57-a089-5395-d436c400418c
* Content: [_NDIS_OFFLOAD_PARAMETERS (ntddndis.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddndis/ns-ntddndis-_ndis_offload_parameters)
* Content Source: [wdk-ddi-src/content/ntddndis/ns-ntddndis-_ndis_offload_parameters.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddndis/ns-ntddndis-_ndis_offload_parameters.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2021-12-01:
Hi @sandilyabhagi. Thanks for opening this issue. We've updated https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddndis/ns-ntddndis-_ndis_offload_parameters with these values.

closed 2021-12-01 by aviviano

sandilyabhagi, 2021-12-01:
Hi @aviviano  Thank you for the update. Is the change is in WDK ? If yes, can you please update the WDK Version.

aviviano, 2021-12-01:
Yes @sandilyabhagi, these macros are defined in the WDK.