MicrosoftDocs/windows-driver-docs-ddi #1216: Access needed for FILE_DISPOSITION_IGNORE_READONLY_ATTRIBUTE

maharmstone, 2021-11-13:
> Specifying FILE_DISPOSITION_IGNORE_READONLY_ATTRIBUTE avoids this return value due to the file being read-only, provided the caller has FILE_WRITE_ATTRIBUTES access to the file (the access that would be required to clear the read-only attribute).

I don't think this is quite right. My test for this succeeds with just DELETE access (https://github.com/maharmstone/btrfs/blob/master/src/tests/delete.cpp#L871). Or does it mean that it checks the SD for FILE_WRITE_ATTRIBUTES, rather than the handle?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d2e4e264-c7f4-2b3e-dfcc-f34a6a64c499
* Version Independent ID: 4dfe047d-1d93-4929-b30b-89122d295a4a
* Content: [_FILE_DISPOSITION_INFORMATION_EX (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/ns-ntddk-_file_disposition_information_ex)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_file_disposition_information_ex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_file_disposition_information_ex.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

billziss-gh, 2021-11-24:
I can confirm the same with Win10 and Win11. The documentation appears to be wrong here.

barrygolden, 2021-12-07:
@maharmstone, thanks for pointing out the incorrect text. It has been removed from the topic. If you have additional questions regarding the use of the FILE_DISPOSITION_IGNORE_READONLY_ATTRIBUTE flag, you can post a question or comment in the following forums:

[Microsoft Q&A: Windows Hardware development](https://docs.microsoft.com/answers/topics/windows-hardware.html)

[OSR Developer Community](https://community.osr.com/)

closed 2021-12-07 by barrygolden