MicrosoftDocs/windows-driver-docs-ddi #1221: FileReparsePointInformation returns STATUS_INVALID_INFO_CLASS on normal directory

maharmstone, 2021-11-30:
This page as it stands is misleading, as it doesn't point out that this only works on the file `\$Extend\$Reparse:$R:$INDEX_ALLOCATION`. See page 85 of [MS-FSA] (https://winprotocoldoc.blob.core.windows.net/productionwindowsarchives/MS-FSA/[MS-FSA].pdf).

I think https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_file_objectid_information and https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_file_quota_information need to be updated too.

Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 29c1d242-b553-09fb-022d-a8e8d128d342
* Version Independent ID: c6f8cd2a-612e-38af-a60b-7f9538e2980c
* Content: [_FILE_REPARSE_POINT_INFORMATION (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_file_reparse_point_information)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_file_reparse_point_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_file_reparse_point_information.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2021-12-02:
@maharmstone , thanks for filing this issue. The information regarding the special directory was actually documented on _some_ - but not _all_- XxxQueryDirectoryFile pages, which is the better place since this structure isn't used to return the error code. I've updated all of these function pages to be consistent. What I actually did was put the list of possible values for FileInformationClass on [NtQueryDirectoryFileEx ](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ntquerydirectoryfileex) and changed all other similar pages' FileInformationClass params to have a link to it. I also added the STATUS_INVALID_INFO_CLASS to the list of possible error codes on NtQueryDirectoryFileEx (and added a link to the page on all other like pages' Return Value sections). This will avoid out-of-sync issues in the future.

closed 2021-12-02 by lorihollasch

maharmstone, 2021-12-02:
Thanks Lori