MicrosoftDocs/windows-driver-docs-ddi #1226: FileSystem Minifilter limitation when AccessMode is KernelMode

SJKU, 2021-12-06:
**If the AccessMode parameter is KernelMode, the requested access is always allowed. If AccessMode is UserMode, the requested access is compared to the granted access for the object.**

In the description of this function it is mentioned that when AccessMode is KernelMode, the request access is always granted. So even in the minifilter driver's POB_PRE_OPERATION_CALLBACK callback when some of the accesses are denied by the driver, those will be ignored by the OS and requested access will be granted? Could you please clarify

lorihollasch, 2021-12-06:
@SJKU , can you clarify which pages you want clarity on - please provide a link to the page for this issue and for issue #1227 .

SJKU, 2021-12-07:
Hi Iorihollasch,

This is link for the page
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-obopenobjectbypointer

At the bottom of the page it is mentioned that 

**If the AccessMode parameter is KernelMode, the requested access is always allowed. If AccessMode is UserMode, the requested access is compared to the granted access for the object.**

When I tested with a minifilter driver that is masking off some of the access mask, it was not honored by the OS and the new returned handle has access that it has asked.

lorihollasch, 2021-12-08:
@sjku, we're still having a bit of difficulty parsing your issue. Based on what you posted on Q&A (which I discovered while researching this issue), the dev team is wondering whether you called FltSetCallbackDataDirty after amending the contents of the callback data structure. See my answer on Q&A, where I also posted additional resources for you if you are somewhat new to minifilter development. I'm going to close this issue out since it seems to not be a doc issue - feel free to reopen if you still feel the docs need updating.

SJKU, 2021-12-08:
Hi Iorihollasch,

I am not talking about preoperation or postoperation callback in the minifilter driver. I have registered for OB_CALLBACK_REGISTRATION in my filesystem filter driver class that is an "ActivityMonitor" category. So my driver has registered prehandle create callback MydriverPreHandleCreateCallback, so this function will be invoked when any kernel component calls the obopenobjectbypointer function.

ProcCreateOperation[0].PreOperation = MydriverPreHandleCreateCallback;

ObOpenObjectByPointer -> ObpCreateHandle -> ObpCallPreOperationCallbacks -> MydriverPreHandleCreateCallback

NTSTATUS ObOpenObjectByPointer(
  [in]           PVOID           Object,
  [in]           ULONG           HandleAttributes,
  [in, optional] PACCESS_STATE   PassedAccessState,
  [in]           ACCESS_MASK     DesiredAccess,
  [in, optional] POBJECT_TYPE    ObjectType,
  [in]           KPROCESSOR_MODE **AccessMode**,
  [out]          PHANDLE         Handle
);

In my driver callback I am modifying the content of the POB_PRE_OPERATION_INFORMATION parameter for example removing PROCESS_CREATE_THREAD permission. So I am expecting that the handle that is returned to the caller at the end of the ObOpenObjectByPointer() will not have PROCESS_CREATE_THREAD permission. But when the AccessMode parameter of the obopenobjectbypointer is KernelMode, whatever my driver has modified that was not reflected when the handle returned to the caller. When I check the handle permission at the end of the ObOpenObjectByPointer function, the handle has PROCESS_CREATE_THREAD permission. 

So I need clarification on this. Why even if the OB_CALLBACK_REGISTRATION callback modifies the content of the POB_PRE_OPERATION_INFORMATION parameter that was not honored by the OS.

lorihollasch, 2021-12-08:
@sjku, I assumed you were the author of [this question on Q&A](https://docs.microsoft.com/en-us/answers/questions/650036/access-granted-when-obopenobjectbypointer-request.html). Did you call [FltSetCallbackDataDirty](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltsetcallbackdatadirty) after modifying the content of POB_PRE_OPERATION_INFORMATION?

lorihollasch, 2022-01-25:
@SJKU , sorry for the delayed response. I have confirmed with the dev team that the behavior is expected and the documentation was correct: If AccessMode=KernelMode, the OS always allows the requested access, regardless of previously set limitations. I added your situation as a "for example" to the page.

closed 2022-01-25 by lorihollasch

SJKU, 2022-01-25:
Thanks Iorihollasch. Just to convey this the development team. Granting always access creates an issue. For example. in my filter driver, I add some restriction to the handle in the POB_PRE_OPERATION_CALLBACK callback. Any third party code with MS digitally signed driver can use KERNEL_MODE and gets the full access to the handle.