MicrosoftDocs/windows-driver-docs-ddi #1234: HID related docs should be in one place

DJm00n, 2021-12-22:
I found that some HID related docs are lying in [MicrosoftDocs/sdk-api](https://github.com/MicrosoftDocs/sdk-api) repo (https://docs.microsoft.com/en-us/windows/win32/api/_hid/) and some in this repo (https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/_hid/).

Please move them into one place (this repo?) - with corresponding `dinputd`, `ntddkbd` and `ntddmou` folders (). It should help a lot with trying to make sense of the variety of available APIs.

Thanks!

mhopkins-msft, 2022-01-05:
Hi @DJm00n,

Thank you for your feedback about the documentation. 

These APIs are organized based on the development kit to which they belong, the Windows SDK and the Windows Driver Kit (WDK). The documentation teams don't have any control over how these are arranged. We're unable to combine these separate repos.

As always, thank you for caring about our docs.

 ~ *Mark*

closed 2022-01-05 by mhopkins-msft

DJm00n, 2022-01-06:
@mhopkins-msft thank you for response!

I think than then there should be cross references between those parts of docs. Can you do that?

mhopkins-msft, 2022-01-11:
@DJm00n 

I've added crosslinks between those two topics. The updates should be published by tomorrow.

Thanks again!